pollen-robotics/reachy2-sdk #382: Stuck in turn_on()

glannuzel, 2024-09-26:
Sometimes we get stuck in the turn_on() function using jupyter notebooks.
We have to restart the Kernel.

(Not sure this is an sdk issue, but this is a reminder)

glannuzel, 2024-10-24:
Still exists

ClaireHzl, 2025-01-13:
Every time after 10 connections to the SDK, the stub is not working anymore, the turn_on stays stuck in self._stub.SetSpeedLimit(req) then self._stub.TurnOn, etc.

Seen on this code : 
```py
from reachy2_sdk import ReachySDK
import time

ite = 0

while True:
    print("Test ", ite)
    print("connexion")
    reachy = ReachySDK(host='localhost')
    print('turn on')
    reachy.turn_on()
    print('end of turn on')
    time.sleep(1)
    if ite%2==0:
        reachy.goto_posture('elbow_90', duration = 1, wait = True)
    else :
        reachy.goto_posture('default', duration = 1, wait = True)
    print('goto ok')
    reachy.turn_off()
    print('reachy turned off')
    reachy.disconnect()
    time.sleep(1)
    ite += 1
```

closed 2025-01-15 by ClaireHzl