pollen-robotics/reachy2-sdk #465: Only one arm activates during turn-on

Drenayaz, 2024-12-17:
Sometimes, when sending a turn-on command, only one arm activates, and the log shows: "l_arm is off. Goto not sent."

ClaireHzl, 2025-01-13:
test with this script : 
```py
from reachy2_sdk import ReachySDK
import time

ite = 0

while True:
    reachy = ReachySDK(host='localhost')
    reachy.turn_on()
    if ite%2==0:
        reachy.goto_posture('elbow_90', wait = True)
    else :
        reachy.goto_posture('default', wait = True)
    reachy.turn_off()
    reachy.disconnect()
    time.sleep(1)
    ite += 1
```

[image]

The commands of speed limit are well sent : 
[image]

Also, every time after 10 connections to the SDK, the stub is not working anymore, the turn_on stays stuck in self._stub.SetSpeedLimit(req) then self._stub.TurnOn, etc.

ClaireHzl, 2025-01-23:
Can happen on single part too : 

[image]